fairscope/PlanktoScope #225: v2023.9.0-beta.0 Node-RED dashboard is missing camera type after restart

ethanjli, 2023-09-13:
**Describe the bug**
After restarting the Node-RED dashboard without also restarting the backend hardware controller, the Node-RED dashboard no longer knows what type of camera is in the PlanktoScope.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to http://home.pkscope/admin/cockpit/system/services#/nodered.service and log in.
2. In the menu icon (which looks like a vertical ellipsis), restart the service.
3. Go to the "System Monitoring" page of the Node-RED dashboard. Confirm that the "Camera Name" field is empty.
4. On the "Administration" page of the Node-RED dashboard, restart the hardware controller.
5. Return to the "System Monitoring" page of the Node-RED dashboard. Confirm that the "Camera Name" field is no longer empty.

**Expected behavior**
After restarting the Node-RED dashboard, it should be able to determine the camera type and optical characterization information without having to first restart the hardware controller.

ethanjli, 2023-09-13:
Because the ability to restart the Node-RED dashboard is buried in the Cockpit GUI (which is currently listed as being for advanced users), it is not yet considered an officially-supported functionality. This, this issue should not be encountered by most users. However, we will need to be careful about this issue if we instruct users to restart the Node-RED dashboard.

ethanjli, 2023-09-14:
Copying in my analysis from https://github.com/PlanktoScope/PlanktoScope/issues/210#issuecomment-1716711239:

> Ultimately, a proper fix to this issue would be to allow the Node-RED dashboard to query the state of the hardware controller. But the MQTT API is not well-suited to this, though it is possible. Ideally we'd have a REST API for that kind of thing, but we may still need some intermediate solution with our MQTT API.

In the meantime, a reasonable workaround could be to add a message somewhere visible in the Node-RED dashboard indicating that the hardware controller (or the PlanktoScope as a whole) needs to be restarted.

ethanjli, 2023-10-19:
Note: this issue will also interact with #190, because (unless we modify the behavior of the MQTT API or the Python backend for reporting camera type) we won't be able to guarantee that the Node-RED dashboard is ready to receive MQTT messages by the time the Python backend sends its MQTT message reporting camera type.

We could have a fine-grained request/response protocol for each MQTT message type. Or we could just have a request from Node-RED at startup asking the Python backend to dump its entire state (including camera type) over MQTT.

ethanjli, 2024-03-14:
Since @BiyouS is gradually working towards a complete rewrite of the Node-RED dashboard which might solve this problem, I will unassign myself from this issue.

ethanjli, 2024-03-26:
https://github.com/PlanktoScope/device-backend/pull/19 should make it relatively easy to add an API route for querying the camera type. That should probably be addressed as part of redesigning the backend API, so I will not assign myself to it at this time.

ethanjli, 2024-06-05:
In the [2024-04-11 software meeting](https://docs.google.com/document/d/1Iba0_9yNFEDdp1Qy8IYlLP43Ga0DnDr3CdavWV0QGZY/edit#heading=h.e3f0mm3ng2sj), we decided that for now we will try to add request/response API routes to the MQTT API, and for now we will not add any HTTP API routes to the Python backend.

Update: I've just now suggested to @melissadjadoun that this issue could be a task she can work on (specifically, prototyping an MQTT API endpoint for requesting information about the camera type/name, which the Node-RED dashboard can use)

melissadjadoun, 2024-07-17:
To address the problem of the unknown camera after restarting Node-RED, I propose the following MQTT topics and payload structures:

        Topic: camera/info/request
        Payload: { "action": "get_camera_name" }
        
        Response Topic: camera/info/response
        Response Payload: { "camera_name": "HQ Camera" }

OR:
      
        Alternative Topic: camera/name/request
        Payload: { "request": "camera_name" }
        
        Response Topic: camera/name/response
        Response Payload: { "camera_name": "HQ Camera" }

ethanjli, 2024-07-18:
Good start! What I like about your first idea is that it continues to use the `action` field, to stay consistent with the structure of existing MQTT commands. I also like that we don't repeat names as much in your first idea - by contrast, in your second idea we have "camera name" in both the topic (as `camera/name`) and the payloads of commands (as `camera_name`) and responses (as `camera_name`). But in both of your ideas there's some repetition with the word `request` between topics and command payloads.

Building on your ideas, I think we can simplify your first idea as follows:
```
    Topic: camera/info
    Payload: { "action": "get" }
    
    Response Topic: camera/info
    Response Payload: { "camera_name": "HQ Camera" }
```
In the future, we can add additional fields to the response payload, e.g. image sensor name, image field-of-view dimensions (number of pixels long, number of pixels tall). Note that I've kept your `camera_name` response payload field as `camera_name` instead of shortening it to `name`, because each camera has at least two names associated with it (e.g. the "HQ Camera" can also be identified by its sensor name, "IMX477"). Also, this reduces the number of changes needed in the existing Node-RED dashboard, which expects the "camera_name" field.

I like your use of `camera/info` as the base path for request and response topics, but until we refactor the existing MQTT API to match that pattern, I think it would be better for us to follow the same topic and response payload structure as the existing MQTT API. In other words, we should keep things consistent across the API. This will also reduce the number of changes we need to make to the existing Node-RED dashboard, whose implementation assumes that all responses are sent under a `status/` topic. So for now I think we should use:
```
    Topic: camera/info
    Payload: { "action": "get" }
    
    Response Topic: status/camera/info
    Response Payload: { status: "success", "camera_name": "HQ Camera" }
```
Does this sound reasonable to you, and can you think of any potential problems with this?

melissadjadoun, 2024-07-23:
I have adapted my code to your proposal regarding the topics to use.

chevreuill3000, 2026-11-11:
This was a defect of the v2 Node-RED dashboard, whose System Monitoring page showed a camera name that the hardware controller announced only once. That page and that field no longer exist in the current dashboard, nor in the pending interface rework, which replaces the Node-RED dashboard. The general need, an interface that shows the real state of the instrument after any restart, is tracked in #933; please reopen if a value that depends on the camera is still missing after a restart.

closed 2026-11-11 by chevreuill3000