fairscope/PlanktoScope #245: v2023.9.0-beta.1 has weird interactions between Cockpit and GPS for adjusting system time

ethanjli, 2023-09-26:
**Describe the bug**
In the `#6-dev-software` channel on the PlanktoScope Slack, Ana Fernandez Carrera [filed](https://planktoscope.slack.com/archives/C01V5ENKG0M/p1695228363994589?thread_ts=1695049296.153819&cid=C01V5ENKG0M) a test report (attached at the end) in which the GPS with a 3D fix set a correct date and time of Sept 14 2023 at 7:25 PM in the Node-RED backend (though of course the backend log files still had names with incorrect timestamps, as described in #218); then when Ana tried to manually change the system time in Cockpit to Sept 20 2028 at 4 PM, she found that it changed the system time but didn't change the date, resulting in Cockpit showing a system time of Sept 14 2023 at 4 PM. When Ana restarted the Python hardware controller, it created a new log at Sept 14 2023 at 4:04 PM; and when she attempted to use Cockpit to upgrade system packages with APT, APT complained that the distro release files were 5 days and 18 hours in the future (which would be consistent with the system time being at Sept 14).

This suggests that the GPS module isn't actually updating the system clock even when it has a fix and is sending the time to the Node-RED dashboard. It's also possible that the PPS kernel module is doing something weird to prevent the system time from being changed by 5 years, and maybe even by more reasonable increments (this was not described in Ana's test report, but it seems likely to me that she tried a smaller change before trying to change the year to 2028)

**To Reproduce**
I am not yet able to try to reproduce this behavior due to lack of GPS modules.

**Expected behavior**
When a GPS fix is acquired, it should update the system time correctly. When a GPS fix is not possible, it should be possible to use Cockpit to fix the system time. When a GPS fix is acquired but a user tries to use Cockpit to change the system time to something incorrect, I don't know what the expected behavior should be.

**Screenshots**
Screenshots can be found in Ana's test report, which I have also uploaded here:
[Trial_Planktoscope_v2023.09.0-beta.1_180923_afc.pdf](https://github.com/PlanktoScope/PlanktoScope/files/12731511/Trial_Planktoscope_v2023.09.0-beta.1_180923_afc.pdf)

ethanjli, 2023-11-22:
Note: it was decided in the [2023-10-19 software meeting](https://docs.google.com/document/d/1fW8OfzK9r2wZVe9-b4BBmfvQma5U5Fneso1usAfge6k/edit#heading=h.po75ra2ttx26) that we will allow the v2023.9.0 software to be released without fixing this issue, if this is the last issue blocking the release. Currently there are other blocking issues, so I won't remove it from the v2023.9.0 software milestone yet. But I will probably move it out of the milestone eventually.

closed 2026-11-09 by chevreuill3000

chevreuill3000, 2026-11-09:
Closing as no longer relevant: GPS support has been removed from the software, and v2023.9.0-beta.1 has long been superseded.