fairscope/PlanktoScope #249: v2023.9.0-beta.1 Impossible to configure White Balance

pkostyrka, 2023-10-10:
**Describe the bug**
1/ When changing the values for White Balance, the WB does not change. 
2/ It is also impossible to write completely a new value, as the software automatically detect it as "no value".

**To Reproduce**
1/ Trying to change the WB values manually or to use the auto WB imply no change in the WB.
2/ Erase the value to write a new one will conduct to an error, and reset to the old value.

**Expected behavior**
1/ The WB should change.
2/ It should be possible to erase a value to write a new one.

ethanjli, 2023-10-17:
Thank you so much for testing out v2023.9.0-beta.1 and filing this bug report! I am hoping that https://github.com/PlanktoScope/PlanktoScope/pull/238 will help to fix this issue, and that pull request will be included in v2023.9.0-beta.2 (which is not yet released, but hopefully will be released soon). I will add a reply to this issue once v2023.9.0-beta.2 is available for testing.

ethanjli, 2023-12-02:
The v2023.9.0-beta.2 prerelease is now published at https://github.com/PlanktoScope/PlanktoScope/releases/tag/software%2Fv2023.9.0-beta.2 . Please let me know in the next two weeks if this issue is still a problem for you, or if it now behaves as you expect. I would like to close this issue in the next one or two weeks, but let me know if you need a bit more time for testing.

ethanjli, 2023-12-21:
@pkostyrka I hope the workshop went well! I just wanted to check again to see if you had an opportunity to test v2023.9.0-beta.2

pkostyrka, 2023-12-21:
Hello Ethan, sorry for the late answer. It's perfect now !

closed 2023-12-21 by ethanjli