fairscope/PlanktoScope #253: Outdated documentation for expert setup.

hankertrix, 2023-10-23:
The link to the file for the Planktoscope software distribution no longer exists. This [link](https://github.com/PlanktoScope/PlanktoScope/archive/refs/heads/stable.zip) returns a 404.

It is under the section about installing the Planktoscope software distribution:
https://planktoscope.github.io/PlanktoScope/software/expert_setup/

ethanjli, 2023-10-24:
Thanks for reporting this issue! As you noted, the URL (https://github.com/PlanktoScope/PlanktoScope/archive/refs/heads/stable.zip) mentioned in the instructions at https://planktoscope.github.io/PlanktoScope/software/expert_setup/#install-the-planktoscope-software-distribution is currently invalid. This is because I wrote those instructions so that they would be usable once we release the next version of the PlanktoScope software, v2023.9.0 (which is when I'll make a new "stable" branch of the http://github.com/PlanktoScope/PlanktoScope repository, so that the https://github.com/PlanktoScope/PlanktoScope/archive/refs/heads/stable.zip URL will work). We are currently in beta testing for the next stable release, but it is taking longer than planned because this is my first time managing a release of the PlanktoScope software and I made some planning mistakes. If you would like to test out the "Expert Setup" process in the meantime, you can use the following commands instead:
```
cd /home/pi
wget https://github.com/PlanktoScope/PlanktoScope/archive/refs/heads/master.zip
unzip master.zip
rm master.zip
mv PlanktoScope-master /home/pi/PlanktoScope
```
This will set up the latest development version of the software, which is basically the same as v2023.9.0-beta.1 but with a few bugs fixed. If you try this, please be aware of the known problems listed for the v2023.9.0-beta.1 prerelease, near the bottom of https://github.com/PlanktoScope/PlanktoScope/releases/tag/software%2Fv2023.9.0-beta.1 . Also, I myself am currently encountering some errors with the expert setup process - starting from yesterday I started seeing hash mismatch errors in the setup process's installation of Python dependencies from https://www.piwheels.org/ - so the expert setup process might not work for you until I fix those problems. I will add a reply to this issue once I get it working again.

I apologize about the confusing state of the docs right now, where the Expert Setup page contains instructions which don't work yet - https://planktoscope.github.io/PlanktoScope is a new version of our docs site for the new v2.5 PlanktoScope hardware and for the v2023.9.0 software (meant to replace https://planktoscope.readthedocs.io/en/latest/ , which only describes the v2.3 PlanktoScope hardware and v2.3 PlanktoScope software). My intention was to launch the new docs site with the v2023.9.0 software release, but there was some confusion which led to the new docs site already being listed on [planktoscope.org](http://planktoscope.org), and I have not yet managed to implement a way to make things clearer.

I will leave this issue open until the software's v2023.9.0 stable release, at which point the "Expert Setup" instructions should work correctly.

hankertrix, 2023-10-25:
You're welcome!

I see. I tried setting up using the master branch of the repository but I encountered issues with docker not being found in the raspbian repositories.

[image]

Ah I see. However, the expert setup on the v2.3 PlanktoScope hardware also doesn't work as the repository has changed a lot and there's no longer a `requirements.txt` file to install from.

<br>
<hr>
<br>

My main issue is that I want to be able to test the installation and the wiring, which I now cannot do because the expert setup does not work anymore. Trying to SSH into the Raspberry Pi that is running the Planktoscope image results in a login screen that I cannot get past as I don't know the username and password to log in to the Planktoscope.

[image]

ethanjli, 2023-10-25:
> I encountered issues with docker not being found in the raspbian repositories.

Oh, interesting - are you using the latest version of the Raspberry Pi OS (bookworm) which was released this month? If so, it sounds like you've discovered that the PlanktoScope OS setup scripts don't yet work on Raspberry Pi OS bookworm - the setup scripts were written for bullseye, the version last updated in May. If this is the situation, the docs are outdated, and I will update them to reflect that the setup scripts must be used with Raspberry Pi OS bullseye, not bookworm. You should instead use one of the following Raspberry Pi OS SD card images:

- https://downloads.raspberrypi.com/raspios_armhf/images/raspios_armhf-2023-05-03/2023-05-03-raspios-bullseye-armhf.img.xz (Raspberry Pi OS bullseye with desktop)
- https://downloads.raspberrypi.com/raspios_full_armhf/images/raspios_full_armhf-2023-05-03/2023-05-03-raspios-bullseye-armhf-full.img.xz (Raspberry Pi OS bullseye with desktop and recommended software)
- https://downloads.raspberrypi.com/raspios_lite_armhf/images/raspios_lite_armhf-2023-05-03/2023-05-03-raspios-bullseye-armhf-lite.img.xz (Raspberry Pi OS bullseye Lite)

The official PlanktoScope SD card images use Raspberry Pi OS Lite, and they don't include a graphical desktop or web browser which would be needed to access the PlanktoScope software's graphical user interface directly from the Raspberry Pi. For this reason, you might prefer to use the Raspberry Pi OS with desktop instead - and from the first photo you posted, it looks like this is what you're doing.

> the expert setup on the v2.3 PlanktoScope hardware also doesn't work as the repository has changed a lot and there's no longer a requirements.txt file to install from.

Yes, if you wanted to try to use the v2.3 PlanktoScope software expert setup process, you would need to use a v2.3 copy of the repository (https://github.com/PlanktoScope/PlanktoScope/archive/refs/tags/v2.3.tar.gz). However, I don't recommend trying to do expert setup with the v2.3 software because that process is very complicated (since it's fully manual), and I had trouble with various parts of it when I was learning the process in order to make it automated for v2023.9.0.

> My main issue is that I want to be able to test the installation and the wiring, which I now cannot do because the expert setup does not work anymore. Trying to SSH into the Raspberry Pi that is running the Planktoscope image results in a login screen that I cannot get past as I don't know the username and password to log in to the Planktoscope. I have tried using the Raspberry Pi default username (pi) and password (raspberry) as well as the username (pi) and password (copepode) found in the documentation, but they both don't work.

Hmm, interesting. On the official PlanktoScope SD card images, the username is always `pi` and the password is always `copepode`.

My first follow-up question for troubleshooting is what keyboard layout is on your keyboard - if you can post a photo of your keyboard, that would help. If you download the v2.3 SD card image from https://github.com/PlanktoScope/PlanktoScope/releases/tag/v2.3, it uses a non-US keyboard layout (either a British layout or a French layout - I think it's the British layout but I forget); if you download the v2023.9.0-beta.1 SD card image from https://github.com/PlanktoScope/PlanktoScope/releases/tag/software%2Fv2023.9.0-beta.1, that uses a standard US keyboard layout. My intuition is that in either case, if you are also using a US-layout keyboard, the `pi`/`copepode` username and password should still work though. If your first goal is to test the PlanktoScope hardware and wiring, I would recommend using one of those SD card images rather than trying to use the Expert Setup process.

Based on the second photo you posted, it looks like you have plugged in the Raspberry Pi to a monitor and keyboard, and you're trying to log in that way with one of the official PlanktoScope SD card images (I believe you're trying to use a v2.3 SD card image?), rather than logging in to the PlanktoScope from another computer over SSH. Even if the `pi`/`copepode` username and password worked for you here, logging in this way wouldn't be very useful for testing the PlanktoScope hardware and software on the official PlanktoScope SD card images. This is because the PlanktoScope software must be controlled with a web browser, but the official PlanktoScope SD card images don't provide a graphical desktop environment or web browser for access over a keyboard and monitor connected directly to the Raspberry Pi. Instead, you should have another computer connect directly to the Raspberry Pi either with an Ethernet cable or over a wifi network which the Raspberry Pi generates as a local hotspot (the name of the wifi network should start with either "PlanktoScope" on the v2.3 software or "pkscope" on the v2023.9.0 software). Then you can try to access the PlanktoScope's graphical user interface from your web browser (either http://planktoscope.local:1880/ui on the v2.3 software or http://planktoscope.local on the v2023.9.0 software), and try to connect over SSH to user `pi` at host `planktoscope.local` (on port 22) with password `copepode`. I recommend trying this out and posting screenshots of what you see, and of any error messages you see.

Another thing you can do to troubleshoot the problem of `pi`/`copepode` not working for you as the username and password is to use the [Raspberry Pi Imager](https://www.raspberrypi.com/software/) program to flash an official PlanktoScope SD card imager to your SD card with a custom password and (if necessary) a custom keyboard layout, both configured in its "Advanced Settings"/"OS Customization" interface (more details [here](https://9to5linux.com/raspberry-pi-imager-gets-new-tabbed-os-customization-ui-raspberry-pi-5-support)). I am interested to see what behavior, and what error messages (if any) you encounter this way.

hankertrix, 2023-10-26:
> Oh, interesting - are you using the latest version of the Raspberry Pi OS (bookworm) which was released this month?

Yeap I am using bookworm.

> The official PlanktoScope SD card images are based on Raspberry Pi OS Lite, so they don't include a graphical desktop or web browser which would be needed to access the PlanktoScope software's graphical user interface directly from the Raspberry Pi. For this reason, you might prefer to use the Raspberry Pi OS with desktop for the "Expert Setup" process - and from the first photo you posted, it looks like this is what you're doing.

> Yes, if you wanted to try to use the v2.3 PlanktoScope software expert setup process, you would need to use a v2.3 copy of the repository [...]

I see, I'll keep that in mind, thanks!

> My first follow-up question for troubleshooting is what keyboard layout is on your keyboard - if you can post a photo of your keyboard, that would help. If you download the v2.3 SD card image from [...]

My keyboard layout is a standard US-layout keyboard, but since you recommended not using the expert installation to test the installation and the wiring, I'll most likely not proceed with using the expert setup.

> Another thing you can do to troubleshoot the problem of pi/copepode not working for you as the username and password is to use the [Raspberry Pi Imager](https://www.raspberrypi.com/software/) [...]

I will keep this in mind, thanks!

<br>
<hr>
<br>

> Then you can try to access the PlanktoScope's graphical user interface from your web browser (either http://planktoscope.local:1880/ui on the v2.3 software or http://planktoscope.local/ on the [...]

I have done that, but the error messages don't give me much information as to what has gone wrong:

[image]

Is there a way to make the interface display more information about the errors?

ethanjli, 2023-10-26:
> I have done that, but the error messages don't give me much information as to what has gone wrong:
Based on the screenshot you posted, it looks like you've managed to have your browser access something on the PlanktoScope, but your web browser has accessed the software editor for the graphical user interface, rather than the graphical user interface itself! Can you post a screenshot of your entire browser window, at least including the navigation bar which shows the URL you're currently on?

It looks like you're probably using the v2.3 software - in that case, you should be trying to open http://planktoscope.local:1880/ui instead of  http://planktoscope.local:1880 (which is what I think you might have opened).

ethanjli, 2023-10-30:
Great! I ended up making changes in #256 to improve the documentation in response to items raised in this issue:
- "Expert Setup" commands didn't work because the stable branch didn't exist: now the "Expert Setup" commands default to the latest beta release, and the instructions specify how to adjust those commands to use the latest stable release instead
- Additional clarification about not being able to operate the PlanktoScope software from a display directly connected to the Raspberry Pi

So I will close this issue as done now.

If you encounter additional problems, please feel free to [create one or more Q&A discussion items](https://github.com/PlanktoScope/PlanktoScope/discussions/new?category=q-a) in our GitHub Discussions board; and then we can break out any specifically identified software/documentation bugs into separate GitHub Issues items 😃

closed 2023-10-30 by ethanjli

hankertrix, 2023-10-31:
Okay, that's great! Thank you!