fairscope/PlanktoScope #257: Optic configuration preview screen blank/black

fishballcheeks, 2023-10-26:
Optic configuration preview screen blank/black (refer to image below)
[image]
Accessed the back-end GUI with http://192.168.4.1:1880/ (refer to images below), however, am unable to attain further details about the errors flagged.
[image]
[image]
I suspect there might be a problem with the Adafruit Stepper Motor HAT as the module controls 3 steppers : 2 for the focus and 1 for the pump stepper motors, please advise.

ethanjli, 2023-10-26:
Hi - thanks for posting this issue! Can you please post a screenshot of the entire "Optic Configuration" dashboard page? If it doesn't all fit on one page, please zoom out in the browser tab until everything fits. I'd like to understand what other settings are currently active.

The most common cause of seeing a black rectangle where the camera preview should be is that the illumination LED either is off or is not shining at the microscope objective lens.

fishballcheeks, 2023-10-30:
Screenshot of the optic configuration dashboard as requested.
[image]
Photo of the Planktoscope set up.
[image]

ethanjli, 2023-10-30:
From the screenshot and photo, it looks like the software is successfully turning the illumination LED on, but light from the illumination LED is not reaching the camera. I see various issues in the hardware assembly (e.g. imprecise alignment of acrylic pieces, and insufficient alignment of the lenses).

One way to test my hypothesis is that if you partially disassemble the PlanktoScope and shine your phone's camera flashlight LED directly into the Raspberry Pi Camera (not having the light go through either of the PlanktoScope's lenses), you should see something in the camera preview. Please try this out and share a screenshot of what the camera preview shows.

If my hypothesis is correct, you will need to redo your assembly of the acrylic parts so that they exactly match the photos in the following sections of the hardware assembly documentation:

- https://planktoscope.readthedocs.io/en/latest/assembly_guide/#step-4-magnets-setup
- https://planktoscope.readthedocs.io/en/latest/assembly_guide/#step-12-pump-mounting

ethanjli, 2023-11-09:
@fishballcheeks Are you still experiencing problems? Or if everything is working now, what did you change to get it to work?

I will close this issue in a few days if there isn't anything to follow up on. Thanks!

fishballcheeks, 2023-11-09:
Hello Ethan,

I'm currently overseas for conference/workshop and will not be able to work on the Planktoscope until end November, please do not close this issue.

Audrey

fishballcheeks, 2023-11-24:
I have shone my phone's camera flashlight LED directly into the Raspberry Pi Camera as well as through the PlanktoScope's lenses,  I apologise for the lack of a better image.
[image]

I have also done some realignment of the plates, however, the optic configuration screen still appears blank/black.
[image]

ethanjli, 2023-11-25:
The alignment looks good to me!

In the first photo you posted, it appears that the PlanktoScope camera is able to show light (i.e. the screen shows a bright/white rectangle rather than a black rectangle) when you shine your phone's flashlight LED. Just to confirm, did you see the same result both when you pointed your phone's flashlight LED directly into the Raspberry Pi camera, and when you pointed your phone's flashlight LED at the objective lens? If so:

1. Can you try shining your phone's flashlight LED at the objective lens, while positioning the phone at the same place where the PlanktoScope's illumination LED is - what do you see on the screen in this case? (please share a screenshot)
2. Does the screen change from showing a black rectangle to showing a bright/white rectangle whenever you move your phone into position (such that the flashlight LED is shining into the lenses), and then back to a black rectangle whenever you move your phone away from the PlanktoScope? (I want to confirm that the display of the camera preview is not freezing for you)

fishballcheeks, 2023-12-01:
In response to:
1. When I shone my phone's flashlight LED at the objective lens, while positioning the phone at the same place where the PlanktoScope's illumination LED is **I saw a blank screen** unlike when I shone my my phone's flashlight LED directly into the objective lens.
2. The **screen does change** from showing a black rectangle to showing a bright/white rectangle whenever I move my phone.

ethanjli, 2024-01-25:
Hi, @fishballcheeks ! I just wanted to follow up again to see if you've been able to figure out or resolve this issue with the illumination LED somehow not illuminating the objective lens in the way it's supposed to. I'll close this issue as resolved if there's no activity in the next seven days, though we can always reopen this issue if needed.

ethanjli, 2024-02-06:
Since there has been no new activity on this issue in the past few months, and no follow-up to my last reply two weeks ago, I will close it as stale.

To anyone who sees this comment in the future: for any new information on this issue, or for any future support requests, please make a new post in the [Help/Support section](https://github.com/PlanktoScope/PlanktoScope/discussions/categories/help-support) of our GitHub Discussions forum 🙂

closed 2024-02-06 by ethanjli