fairscope/PlanktoScope #259: raspimjpeg captures overexposed images

ethanjli, 2023-10-26:
**Describe the bug**
raspimjpeg seems to ignore or increase the exposure settings (shutter speed, ISO) set on the Optic Configuration page when it actually saves images. This leads to overexposed images. Thibaut has been working around this by figuring out ways to reduce the intensity of the illumination LED (with tape as a hardware workaround, or with LED dimming as a software workaround) in the v2.5/v2.6 hardware, so that at least raspimjpeg's overexposure behavior won't result in excessively washed-out images.

**Expected behavior**
The apparent exposure of acquired images should match the apparent exposure seen in the camera preview.

**Additional context**
Additional context is available in https://github.com/PlanktoScope/PlanktoScope/issues/79#issuecomment-1707470541 , and in the [2023-10-26 PlanktoScope software taskforce meeting notes](https://docs.google.com/document/d/1fW8OfzK9r2wZVe9-b4BBmfvQma5U5Fneso1usAfge6k/edit#heading=h.7qfpl4u25mdo), and in the [2023-10-19 software meeting notes](https://docs.google.com/document/d/1fW8OfzK9r2wZVe9-b4BBmfvQma5U5Fneso1usAfge6k/edit#heading=h.po75ra2ttx26).

Wassim is working on a longer-term fix by migrating from raspimjpeg to picamera2. In the meantime, we may need to include some software workarounds in the mainline software, e.g. by including an LED dimming setting in the pscopehat version of the Python backend.

ethanjli, 2023-10-26:
Note: I'm assigning this issue to myself for implementing any temporary software workarounds identified by Thibaut et al., and I plan to assign this issue to Wassim for the picamera2 migration project.

ethanjli, 2023-11-15:
Following up on https://planktoscope.slack.com/archives/C01V5ENKG0M/p1700060664935139, I am assigning Laurent for changing the default LED supply current in the pscopehat software as a temporary workaround for the issue, and I'm un-assigning myself from this issue.

LaurentPV, 2023-11-29:
We tested today on culture samples and with the led fed by a controlled voltage and current.
A current of 10mA resulting in a voltage of 2,77v seemed to provide better results with the full range of ISO settings usable.
Here are an example with 10mA
[image]

Here an example with 20mA
[image]

ethanjli, 2023-12-01:
I'm temporarily adding this issue to the v2023.9.0 milestone specifically because we need to include https://github.com/PlanktoScope/device-backend/pull/14 in the milestone; once we merge in that PR and update the distro setup scripts to use an updated version of device-backend, then I will move this issue from the v2023.9.0 milestone back to the Backlog milestone.

ethanjli, 2024-03-26:
This PR was fixed by #380, but I had forgotten about the existence of this PR when I had merged #380 - so I am manually closing this issue.

closed 2024-03-26 by ethanjli