fairscope/PlanktoScope #268: The documentation doesn't explain our version numbering system.

ethanjli, 2023-10-26:
**Describe the bug**
Neither the planktoscope.org website nor the docs site clarifies what we mean when we refer to "v2.3 hardware" and "v2.3 software" and "v2023.9.0" software in the same sentence.

**Expected behavior**
New and returning users should have a clear understanding of which version of software to use with which version of hardware, etc.

**Additional context**
This issue was raised by @kitazy52 in the [2023-10-26 planktoscope software meeting](https://docs.google.com/document/d/1fW8OfzK9r2wZVe9-b4BBmfvQma5U5Fneso1usAfge6k/edit#heading=h.5rv7e2bqpld4).

ethanjli, 2023-10-26:
@kitazy52 would you be able to describe in a bit more detail what some of the concrete issues are with the current situation, and the ideas you mentioned during the meeting for how to improve the situation?

kitazy52, 2023-10-30:
@ethanjli I'm sorry for not getting back to you sooner. The history of the versioning is now complicated to recognise for those who have not worked with PlanktoScope for years (e.g.  v2.3 -> v.2023.9.0 -> v.2024.1.0 ???). Considering the current situation on Slack that there are so many misunderstandings in posts, we had better include the version history on the planktoscope.org website or the docs site. (This could be the case both for the hardware and the software)

ethanjli, 2023-10-30:
Thanks for the clarification! I will think about where I can explain the history of version numbering in the docs site.

PR #274 (2023-11-21): Make the docs site easier to navigate
This PR improves the navigability of the docs site by:
- [x] Consolidating the navbar (fixes #263)
- [x] Merging the navbar into the left navigation sidebar (part of #275)
- [x] Restructuring the docs based on what goals users may have when visiting the documentation (part of #275)
- [x] Simplifying the docs site's homepage to help the user get to where they probably need to go (part of #275)
- [x] Hiding the tag of the latest release on GitHub (fixes #267)
- [x] Starting the documentation of post-installation configuration (for hardware settings), as part [... cut]
closed 2023-11-21 by ethanjli