fairscope/PlanktoScope #270: Metadata box in the sample tab for time between sampling and analyze of the sample

pkostyrka, 2023-10-30:
**Is your feature request related to a problem? Please describe.**
Even if you use the bubbler, the sample will settle over time. The time you analyse your sample after collection is therefore important, as it may affect your results. 

**Describe the solution you'd like**
It should be possible to fill in the sampling time and, separately, the time at which the sample is analysed. Even better if it can calculate automatically the difference between the two of them.

pkostyrka, 2023-10-30:
@ethanjli I mean, we should be able to get the time when we process the sample (which is not possible for now as the intern clock of the PlanktoScope is not working). So to avoid this, we could had something similar to the date/time field, but to enter the time we process the sample (and not the sampling time field). Then we could figure out which samples are processed "too late" after the sampling or not.

ethanjli, 2023-10-30:
I'm not sure exactly what you mean by "process the sample", which seems to be a source of my confusion. To check my understanding, here are the various possible events (and thus timestamps) I can think of:

1. Sample collection started (e.g. plankton net goes in the ocean)
2. Sample retrieved (e.g. plankton net comes out of the ocean)
3. Wet-lab sample processing (e.g. filtering the raw sample) started
4. Wet-lab sample processing (e.g. filtering the raw sample) completed
5. Wet-lab-processed sample loaded into the PlanktoScope
6. Image acquisition started
7. Image acquisition completed

The meaning of the "Sampling Time" field we currently have on the "Sample" page varies depending on sample type. With the "Single location" sampling type, it's either timestamp 1 or timestamp 2 (it's left ambiguous, but probably most people will interpret it as timestamp 2); with the "Plankton net" sampling type, the net throw timestamp is timestamp 1 while the net retrieval timestamp is timestamp 2.

Currently, the only way to record timestamps 3-5 would be in an external spreadsheet. Timestamp 3 would enable us to determine the delay between when a sample (e.g. from a plankton net) is poured into a large bucket for storage in a refrigerator and when some small volume of water is taken from the bucket to be filtered for imaging on the PlanktoScope. Usually, when there is a meaningful delay between those events, you should thoroughly mix the bucket before taking water from the bucket, to mitigate the settling of plankton over time - even if you insert a bubbler into the bucket. Or if a long time has passed between the events described by timestamps 4 and 5, then the sample might have settled in the PlanktoScope's sample tube (the syringe body above the flowcell, in the v2.5/v2.6 hardware design) despite the bubbler in the PlanktoScope's sample tube.

Timestamp 6 is automatically recorded (without user-facing action) by the PlanktoScope software using the Raspberry Pi's system clock when the user presses the "Start Acquisition" button on the "Fluidic Acquisition" page. However, the system clock is usually inaccurate in past and current versions of the PlanktoScope software with the v2.5/v2.6 hardware (and any past or future version of the PlanktoScope software with the standard v2.1 PlanktoScope hardware) when the PlanktoScope can't get a GPS signal to fix the system clock, because of issue #95 (where the PlanktoScope OS image isn't properly configured to use the hardware real-time clock module in the v2.5/v2.6 hardware which should always maintain the correct time for the system clock). Recording timestamps 2 and 6 would enable us to determine the delay between when a sample (e.g. from a plankton net) comes out of the ocean and when imaging is started on some plankton from that bucket. Usually we want to minimize this delay because the sample will gradually degrade after it comes out of the ocean. Recording timestamps 5 and 6 would enable us to keep track of which samples might have settled due to a long delay between those two events.

Currently, I believe timestamp 7 is not explicitly recorded as metadata. If the sample settles over the course of image acquisition because the bubbler in the PlanktoScope's sample tube is not powerful enough, then the difference between timestamp 5 and timestamp 6 would allow us to identify whether an image acquisition took too long.

Are you proposing that we have a user-input field for timestamp 3, or a user-input field for timestamp 5, or a user-input field for timestamp 6, or something else? Also, is this feature request intended as a temporary workaround until #95 is fixed, or as something that will be relevant even after #95 is fixed?

pkostyrka, 2023-11-07:
Hello Ethan, 
You're right, it is a temporary workaround until #95 is fixed. So we could have the time for the step 5-6. I am proposing a user-input field for the timestamp 6 (or at least 5). But if the issue #95 is fixed, we will have that data anyway. Thank you for your clarification !

ethanjli, 2023-11-08:
Ok! In that case, I will see if I can fix #95 for the next software release (v2024.0.0) after our upcoming one (v2023.9.0), and I will close this issue as "wontfix".

closed 2023-11-08 by ethanjli