fairscope/PlanktoScope #275: Provide a clear path through the docs for users trying to get started

ethanjli, 2023-10-31:


ethanjli, 2023-11-22:
I'm adding this to the [software v2023.9.0 milestone](https://github.com/PlanktoScope/PlanktoScope/milestone/3) specifically for the "Update the operation documentation for the PlanktoScope's new wifi hotspot naming scheme, URL, and landing page" task, to make sure I don't forget to do it for v2023.9.0. However, the other remaining tasks in this issue are out-of-scope for v2023.9.0, and I will move this issue back into the Backlog milestone afterwards.

ethanjli, 2023-11-30:
Since #294 finishes the part of this issue which needs to be completed for the software v2023.9.0 milestone, I will remove this issue from that milestone.

chevreuill3000, 2026-11-09:
Closing as not planned: there has been no activity on this issue for a long time and it is not on the current roadmap. Feel free to reopen it if it is still relevant.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-11-09 by chevreuill3000