fairscope/PlanktoScope #289: The flowrate is not the real flowrate

pkostyrka, 2023-11-22:
**Describe the bug**
With a 2045 step/ml for the pump with a flowrate of 5 mL/min, when you pass 10 mL, the pump effectively pass 10 mL.
But, it takes around 3 minutes, when it should be 2 minutes. So the real flowrate is around 3.5 mL/min.
(I did this on ten units).

**To Reproduce**
Steps to reproduce the behavior:
1. OPTIC CONFIGURATION
2. Pass 10 mL with a flowrate of 5 mL/min. Measure the time. It should be 2 minutes if the real flowrate is 5 mL/min. But it is not the case.

**Expected behavior**
If you enter a flowrate, it should be the real flowrate.

ethanjli, 2023-11-22:
Which version of the PlanktoScope hardware are you using?

ethanjli, 2023-11-28:
Update: I just realized that I misread your description, so I misunderstood the problem you were reporting. If I understand correctly now, you actually do have a correct calibration for the `pump_steps_per_ml` setting, since the pump will pump exactly 10 mL of fluid when you tell it to pump 10 mL of fluid. Instead, it seems that there is some other setting which will need to be calibrated in order to calibrate the pump flowrate.

pkostyrka, 2023-11-28:
Yes that's right !

ethanjli, 2024-12-13:
Currently, a workaround to adjust the flowrate calibration is to adjust the calculation of the `steps_per_second` variable at https://github.com/PlanktoScope/device-backend/blob/f4d60a32ddff582aad6031c74d9abe8c3c838ff9/control/planktoscopehat/planktoscope/stepper.py#L387C9-L387C69, by adding an extra multiplicative factor. The code (and specifically the flow of values) is rather hard to follow in that file, so it may also be possible to adjust the flowrate calibration by adjusting some other variable/constant in that file - however, Jeremy Szymczak (at FairScope) and I worked to try to adjust various other variables/constants in that file but we didn't see the expected outcomes from those other adjustments.