fairscope/PlanktoScope #290: A Metadata reorganization to facilitate informations sharing

LaurentPV, 2023-11-23:
## Motivation

For now metadata variables are dispatch between two files : config.json and hardware.json.
Some of the hardware variables in config.json and some settings variables in hardware.json.
It is not easy to know where to look when the user search for a metadata in the configuration files.
We would like to facilitate acquisition settings sharing between users, retrieving hardware configuration for Fairscope and prepare new features such as direct export to EcoTaxa or Fairscope server.

## Proposal
We propose to add new metadata to improve data quality, to order them properly so that hardware.json contains only hardware caracteristics and config.json contains only user settings.
We propose to add a personal_info.json file that contains the sample_ metadata and user id so that config.json can be shared to expect same acquisition results and personal_info contains user id and campaign data so that those informations are stored.

A list of metadata is available here : https://docs.google.com/spreadsheets/d/1TSIaOFEIMvvYyqAFrsiZxVtGXZvWVdZbWO_LU-2A_TE/edit?usp=drive_link

ethanjli, 2023-11-23:
One observation: part of the surprising behavior described in #209 may be resolved by thoughtful organization of the metadata fields with the approach described by this proposal. The current version of the design document in #282 ([permalink](https://github.com/PlanktoScope/proposals/blob/d8dfa8f594e18f2c84ea9861cd12097bc22c7590/design/0290-metadata-reorganization.md#proposal)) proposes that the white balance values should be moved from the `hardware.json` file to the `config.json` file. As a result, the white balance values would no longer be reset to hardware-specific defaults when the user a different hardware version in the Node-RED dashboard. This would be a desirable result.

ethanjli, 2023-11-28:
Overall, I agree with the proposal - good work!

I have some comments, suggested changes, and ideas:
- In the `hardware.json` file, you improvised a serialization format for the `_model` fields with a `[manufacturer] - [part number]` format (e.g. `Kamoer - KAS B12 SF`). I wonder if it might be cleaner to have separate fields for information about the manufacturer vs. information about the specific part. So, for example, we might have the following fields:
  - `inst_manufacturer`: `FairScope`
  - `inst_model`: `PlanktoScope v2.6.1`
  - `inst_computer_manufacturer`: `Raspberry Pi` (or `NVIDIA`)
  - `inst_computer_model`: `Raspberry Pi 4 4Gb` (or `Jetson Nano`)
  - `inst_camera_manufacturer`: `Raspberry Pi`
  - `inst_camera_model`: `High Quality Camera`
  - `inst_hat_manufacturer`: `Adafruit` (or `FairScope`)
  - `inst_hat_model`: `Stepper Motor HAT` (or `PlanktoScope HAT v1.3`)
  - `inst_led_manufacturer`: `Shenzhen Furuier Photoelectric`
  - `inst_led_model`: `FLR-50T04-HW7`
  - `inst_pump_manufacturer`: `Kamoer`
  - `inst_pump_model`: `KAS-B12-SF`
  - `inst_flowcell_manufacturer`: `FairScope` (or `Ibidi`)
  - `inst_flowcell_model`: `300 um Capillary` (or `µ-Slide I Luer Uncoated 200µm`)
  
  On the other hand, maybe there is nothing practically useful to us about tracking the manufacturer and model separately. In that case, maybe it is better to just simplify the `_model` fields by removing the ` - ` separator from the value, e.g. `Raspberry Pi 4 4Gb`, `Kamoer KAS-B12-SF`, `FairScope 300um Capillary`, `FairScope PlanktoScope HAT v1.3`, `Shenzhen Furuier Photoelectric FLR-50T04-HW7`, etc.
- The `process_id` metadata field should be removed from `hardware.json`, since that field is only used by the segmenter. That ID should be unique per raw dataset, but we can reuse the same ID (e.g. `1`) for different datasets. Thus, it might not be necessary to persist this value in a JSON file.
- To make it easier to understand the metadata fields in the metadata files, I would like to give identical name prefixes for fields which are logically related.
  - For example, the `microsteps` field should be renamed to either `pump_microsteps` or `focuser_microsteps` depending on which motor the `microsteps` field is used for; and the `acq_pump_model` field should be renamed to `pump_model` (for consistency with `pump_max_speed` and `pump_steps_per_ml`).
  - For example, all settings related to stop-flow imaging (which are meaningless outside the context of stop-flow imaging) should have the same name prefix: `stopflow_step_volume` instead of `acq_volume_interframe`, `stopflow_nb_frames` instead of `acq_nb_frames`, and `stopflow_stop_pause` instead of `sleep_before`. There is interest in a future continuous-flow imaging mode (which would be enabled by use of a camera with a global shutter rather than a rolling shutter), and it would be useful to keep separate metadata fields for the settings in the different imaging modes.
  - Thus, I am proposing that we should not follow EcoTaxa's `acq_`, `object_`, `sample_`, and `process_` prefixing scheme for our own internal names of our metadata fields, since a lot of it doesn't make sense (e.g. `object_camera_gain_red` doesn't make sense since, in the PlanktoScope, it is a property of our camera and not a property of objects identified by the segmenter; so instead that metadata field should just be named `camera_gain_red`). Instead, we should choose names which make the most sense within the PlanktoScope, and then translate the names as part of the export for EcoTaxa.
- I wonder if it might make sense to create a separate metadata file called `calibration.json`, to keep the following calibrations separate from the `hardware.json` and `config.json` files which might be replaced by the user:

  - `pixel_size`: used for size calibration (currently in `hardware.json` in the proposal)
  - `focus_steps_per_mm`: used for focusing motor distance calibration (currently in `config.json` in the proposal)
  - `pump_steps_per_ml`: used for pump motor flowrate calibration (currently in `hardware.json` in the proposal)
  - `camera_gain_red` and `camera_gain_blue`: used for white balance calibration (currently in `config.json` in the proposal)

  This way, the `hardware.json` file would purely describe the configuration of the hardware, and would not need to be modified when the user (re)calibrates their machine.

  Additionally, then the `config.json` file would only have image acquisition settings - so maybe then we could rename that file to `acquisition.json` or `image-acquisition.json`?
    - I think that would be nice because it'd be a more specific and descriptive filename.
    - In this case, it might also be reasonable to move the `acq_id` field to this file from the `personal_info.json` file. However, we might want to avoid overwriting `acq_id` when we overwrite the `acquisition.json` file (e.g. if we want to allow the user to save and load different preset `acquisition.json` files), since `acq_id` needs to be unique.
- Right now the `personal_info.json` file contains no personal information beyond `sample_operator`. Maybe we should rename `personal_info.json` to `sample.json`. 
  - In that case, it might be better to store `sample_operator`, `process_id`, `acq_id`, and `sample_id` in a separate file from `sample.json` - it would be better if we avoided overwriting those values when overwriting the `sample.json` file (e.g. if we want to allow the user to save and load different prese [... cut ...] y.
- We will should ensure that we do not remove or rename any of the metadata fields which were previously included in the metadata `.csv` export file for EcoTaxa, in order to keep our EcoTaxa metadata export files backwards-compatible with respect to metadata in datasets exported by previous versions of the PlanktoScope software. This should be explicitly mentioned in the design document.

I might have more feedback after we we finish this iteration of discussion/edits on the design documents.

ethanjli, 2024-03-21:
While working on https://github.com/PlanktoScope/device-backend/pull/19 I gained more familiarity with the camera settings in our hardware configuration; we will probably want to store more camera settings, and we may also want to change the representation of some values, and we may also want to delete some fields which are unused (e.g. `analog_gain` and `digital_gain`). This is just a note to myself to check the TODOs I left in https://github.com/PlanktoScope/device-backend/pull/19 when I work on the design document for this proposal.

ethanjli, 2025-04-15:
As discussed in https://github.com/PlanktoScope/PlanktoScope/pull/565#issuecomment-2802841080 and https://github.com/PlanktoScope/PlanktoScope/pull/565#issuecomment-2803625433, we might want to move the config files from `/home/pi/PlanktoScope/*.json` into somewhere in `$XDG_CONFIG_HOME` (e.g. `$HOME/.config/PlanktoScope/*.json`) together with any backwards-incompatible schema change caused by implementation of this proposal.