fairscope/PlanktoScope #315: Filling metadata in single location net register them, but they disappear from the sample tab

pkostyrka, 2023-12-21:
**Describe the bug**
When you fill metadata (lat, lon, date, etc.) in the single location mode, at the moment you validate them they disappear from the fields. They are correctly registered (you can check it in the acquisition tab).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'sample'
2. fill metadata in single location mode
3. validate

**Expected behavior**
The metadata should stay on their respective fields and not disappear.

ethanjli, 2024-01-16:
I briefly looked into this problem, and I think this behavior is because these input fields are implemented in as a Node-RED "form" node. To fix it, we'll probably have to reimplement it as a collection of text nodes - which I am not yet sure how to do correctly.

ethanjli, 2024-01-27:
Hmm, I wonder if we can keep the input form node type and just feed submitted values back into the form.

PR #365 (2024-01-27): Repopulate input fields of input forms on Sample page
This PR fixes #315 (and also fixes #314, which - based on the work done in this PR - I have determined is actually just a duplicate of #315) by feeding the outputs of the Node-RED dashboard's "Sample" page's input form nodes back as inputs into those nodes once the respective input forms are submitted.
closed 2024-01-27 by ethanjli