fairscope/PlanktoScope #317: You can change the flowrate during an acquisition

pkostyrka, 2023-12-21:
**Describe the bug**
During an acquisition, you can go in optic configuration and change the flowrate. It creates a bug in the software, and it should not be possible to do that.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch an acquistion
2. Go to optic configuration
3. Change the flowrate

**Expected behavior**
It should not be possible to change the flowrate when an acquisition is launched.

ethanjli, 2023-12-21:
When you change the value of the flow rate input box on the optic configuration page during acquisition, what consequences do you observe? e.g. does the pump start running at the new flow rate?

ethanjli, 2024-04-13:
@melissadjadoun the redesign of the MQTT API should probably make the API return an error when a user attempts to change the flowrate while an image-acquisition sequence is running.

chevreuill3000, 2026-11-09:
Closing as a duplicate of #1079, which covers locking all hardware controls (including the flow rate) while an acquisition is in progress.