fairscope/PlanktoScope #356: Display System Date and Time

kevin-duclos, 2024-01-24:
Although it is not used for the metadata, the system date-time is important because it sets the names of data folders. It is easy to have the wrong system date-time with a device that is not designed to always be connected to the internet, which can make data organization confusing.

Displaying the system date-time somewhere in the interface would solve this problem. The most obvious place for this information is near the "Force Local Clock Update" button under the "System Monitoring" tab.

ethanjli, 2024-01-24:
Thanks for submitting this feature request, and I agree with the requested functionality! In PR #346, I have added two entries to the "Information" panel in the System Monitoring page, displaying the system time on the Raspberry Pi and the time according to your web browser:
[image]

This functionality is available to preview on the master branch (e.g. via the instructions at https://docs.planktoscope.community/setup/software/nonstandard-install/#__tabbed_1_3), and it will also be made available for easier preview on the next alpha prerelease of the PlanktoScope software (timeline TBD)

Does those two entries in the "Information" panel fully address your feature request, or are there any additional improvements (to e.g. UX or documentation) needed?

kevin-duclos, 2024-01-25:
Perfect! Thank you for the quick fix.

closed 2024-01-25 by kevin-duclos

ethanjli, 2024-02-06:
I've published an alpha prerelease (i.e. a development preview with SD card images) at https://github.com/PlanktoScope/PlanktoScope/releases/tag/software%2Fv2024.0.0-alpha.0 , which includes the fix I mentioned in my previous comment. Feel free to test it out and leave feedback if you want to preview the changes I've made to the System Monitoring page!