fairscope/PlanktoScope #358: backend/processing/segmenter: Add a scalebar to each segmented object.

ethanjli, 2024-01-26:
**Is your feature request related to a problem? Please describe.**
Currently, all metadata is reported in units of pixels, and the pixel size calibration is not used to calculate any metadata fields (e.g. width, height, area, perimeter, or equivalent diameter) exported to EcoTaxa. This means that users have to do a mental calculation in order to estimate the actual size of an object.

**Describe the solution you'd like**
@fabienlombard has requested that the PlanktoScope software should add a scalebar as a length-calibration annotation to every segmented object image. This is also the behavior of Flowcam (e.g. as in https://ecotaxa.obs-vlfr.fr/prj/11559):
[image]

I agree with this request, and I also think we should make it easy for a script to remove scalebars for subsequent processing (e.g. to compile into a mosaic). If we have a fixed-height padding (e.g. always 48 px at the top of the image, or always 48 px at the bottom of the image), this would make it easy for a script to crop out the scalebar if needed. But the default for most users will be to just accept the existence of the scalebars everywhere, which seems reasonable to me.

**Describe alternatives you've considered**
One potential alternative instead of having a scale bar is to also calculate and export pixel-size-calibrated versions of the metadata fields (i.e. perform unit conversions from pixels or pixels^2 to um or um^2), but that would add complexity to the software and the traceability/correctness of any metadata values derived from calculations. @fabienlombard [thinks we should](https://planktoscope.slack.com/archives/C01V5ENKG0M/p1706296112580049) discourage the calculation of derived values. I think, at the very least, if any derived values are to be computed, it is reasonable for us to say that it is entirely the user's responsibility to do those calculations and to ensure the correctness of whatever they do - and that we will not make any such calculations in the PlanktoScope software. I think it will be simpler for the PlanktoScope software to just add a scalebar instead of doing anything with unit conversions.

ethanjli, 2024-08-08:
[On Slack](https://planktoscope.slack.com/archives/C015K99AJAE/p1723140182993909?thread_ts=1721986053.838529&cid=C015K99AJAE), @fabienlombard says that the padding for the scalebar should be added as 18 rows of white pixels, and the scalebar should only be shown within those 18 rows of white pixels. Apparently this is the region ignored by EcoTaxa's deep visual feature model.

moi90, 2024-10-15:
I don't know how segmentation is currently implemented. But if it is in Python, you could have a look at MorphoCut's implementation: https://github.com/morphocut/morphocut/blob/loki/src/morphocut/scalebar.py