fairscope/PlanktoScope #371: Running the segmenter separately on a local computer

Mrvy29, 2024-02-14:
Hello all planktoscope users and developers!

When analyzing lots of samples with the planktoscope on a time schedule, especially fresh plankton, we found the segmenter was slowing and creating issues with pictures acquisitions in parallel. We resorted to running the samples and finding another time for segmenting.

We are wondering how it to possibly run the segmenter on a regular computer with the  appropriate software installed.  As I understand it potentially on a python local machine/runtime.

How can we access the segmenter's code? Would it be possible to obtain something like that?

Thanks!

ethanjli, 2024-02-28:
Update: I've put together a prototype of a simple set of commands for bringing up the segmenter on a separate computer, along with everything needed to interact with the segmenter from your web browser, at https://github.com/PlanktoScope/pallet-segmenter . It works on my computer (and I found that segmentation is very fast there!). Before it'll be ready for more general usage, I will wait for more people to test it out and provide feedback (both on whether everything works as expected, and on the quality of the usage instructions) - so for anyone who's interested, please test it out! Note that I've tested it on Linux, but not on macOS or Windows, so things might need to be fixed on the two latter platforms.

Mrvy29, 2024-03-10:
Hi Ethan, thanks ao much for the thorough answers, just got back from sea and am eager to try. I work on a Mac so that will broaden OS tested on. I'll let you know how I get on. In the meantime I've uploaded first test sample data to ECOTAXA and have lots more to ask about, i'll prepare these queries on a separate thread.
M

ethanjli, 2024-03-10:
Great! For any questions you have or anything which is unclear (e.g. due to inadequate documentation), please feel free to create one or more posts in  https://github.com/PlanktoScope/PlanktoScope/discussions/new?category=help-support

ethanjli, 2024-03-12:
Update: I've also created a separate issue (#378) to track some work I plan to do eventually for making it simpler to just run the segmenter as a batch-processing routine on a specified list of datasets from the command-line (rather than having to go through a Node-RED dashboard).

ethanjli, 2024-04-13:
As discussed in the [2024-04-11 software meeting](https://docs.google.com/document/d/1Iba0_9yNFEDdp1Qy8IYlLP43Ga0DnDr3CdavWV0QGZY/edit#heading=h.e3f0mm3ng2sj) (in a discussion about designing the next version of the PlanktoScope's MQTT API), we decided that we will try to make it possible to import the segmenter as a Python library for use in a Jupyter notebook.