fairscope/PlanktoScope #378: backend/processing/segmenter: Enable launching as a script for batch processing

ethanjli, 2024-03-12:


ethanjli, 2024-03-14:
This issue was discussed at the [2024-03-14 software meeting](https://docs.google.com/document/d/1Iba0_9yNFEDdp1Qy8IYlLP43Ga0DnDr3CdavWV0QGZY/edit#heading=h.2yvge3qz8v4g). We decided by consensus that we will not prioritize this item for now; FairScope might assign an incoming intern to work on this, depending on the interests of the interns.

tstilwel, 2024-03-18:
Would it be possible to publish the python script without the MQTT dependencies mentioned above?  For those not in the Planktoscope slack channel.  Thanks!

ethanjli, 2024-03-18:
Thanks for your interest! Here is what was posted on the PlanktoScope Slack:

- From Salima Rafai, a "python script for local segmentation": [stand_alone_segmenter.zip](https://github.com/PlanktoScope/PlanktoScope/files/14641613/stand_alone_segmenter.zip); and "an .exe file with the segmenter": [local_segmenter.exe - Raccourci.zip](https://github.com/PlanktoScope/PlanktoScope/files/14641623/local_segmenter.exe.-.Raccourci.zip)
- In response, Jan Heuschle said: "The python works great. Since tkinter had some problems on my mac,  I had to switch the file dialog to use easygui instead. In case you want to check the file , I upload it here.": 
[local_segmenter_easygui.zip](https://github.com/PlanktoScope/PlanktoScope/files/14641635/local_segmenter_easygui.zip)

Note that I have not attempted to review the code in these attached files, and I have not attempted to run these scripts.

If anyone reading this discussion needs a way to run the segmenter without MQTT and the above files don't work for you, please add a reply comment describing your intended use-case! We use feedback from people running PlanktoScopes to help us (re)prioritize what we need to work on

ethanjli, 2024-12-13:
For the record, Maggi Brisbin expressed interest [on the PlanktoScope Slack](https://planktoscope.slack.com/archives/C019Q3PT5T3/p1733447655422929?thread_ts=1733365044.094969&cid=C019Q3PT5T3) for having a way to perform segmentation from the command-line, without having to use the PlanktoScope GUI.