fairscope/PlanktoScope #425: The hardware documentation isn't up to date.

Guelakais, 2024-06-10:
**Describe the bug**
The current version of the PlanctoScope hardware is even more advanced than V2.6.2_03. This has caused some problems when assembling the kit.

**To Reproduce**
Steps to reproduce the behavior:
1.  Part E isn't actual anymore
2.  In the actual kit are to many m2.5 10mm screws
3.  In the actual kit are to less m2.5 5mm screws

**Expected behavior**
I expected to find the same parts in my kit as in the CAD files.

**Additional context**
I had to get 3 extra 2.5 5mm screws from a friend. This delayed the assembly.

**Planktoscope version (please complete the following information):**
 - Raspberry version: Pi4
 - OS: standard image
 - Stepper motor hat custom hat
 - Version 2.6.2_03

ethanjli, 2024-06-11:
Thanks for reporting this issue! Which documentation page(s) have the outdated information? (URL(s) would be helpful)

ethanjli, 2024-06-11:
Thanks, that's the URL I needed you to confirm. For context, our documentation situation is a bit messy right now because we have v2.5 hardware assembly instructions available at https://docs.planktoscope.community/setup/hardware/v2.5/ (and that documentation website is deployed from this git repo, and I maintain the structure and organization of that documentation website, as well as software-related documentation on that site), while the v2.6 hardware assembly instructions at https://docs.google.com/presentation/d/1D8e0fmuGJ5oLSN9MLBFf6NgO67L-fk7eb9GNnxKbqnc/edit?usp=drivesdk are maintained separately by FairScope and have not yet integrated into the docs.planktoscope.community site. I will pass this issue report along to @tpollina / FairScope for you, and I'll post updates here!

ethanjli, 2024-06-11:
Here's what I heard from Tanguy Cebron at FairScope:
> Part E as a change of small design (external shape) but it doesn't change the assembly procedure at all. It can be updated in next verison.
> 
> For the screw it might be a mistake in picking if there a screw exchange. But design doesn't change on that side. How many of each was in the kit?

Feel free to reply to Tanguy's follow-up question [directly on Slack](https://planktoscope.slack.com/archives/C014353MWHY/p1718118878791679?thread_ts=1718117754.221479&cid=C014353MWHY), or to reply here (in which case I'll continue to relay replies between Slack and this issue)

ethanjli, 2025-03-20:
Tanguy Cebron (@tcebron ) has been working on documenting the v2.6 hardware assembly process, so I'm transferring responsibility for handling this issue to him.

closed 2026-11-12 by chevreuill3000