fairscope/PlanktoScope #426: current cad files cannot be viewed properly.

Guelakais, 2024-06-10:
**Describe the bug**
I had tried to open the .f3z files to view the current cad files. On the one hand, this was not easily possible because fusion did not accept the f3z file. On the other hand, the unzipped f3d files could only be opened partially.
**To Reproduce**
Steps to reproduce the behavior:
1. try to download the current case [file](https://github.com/PlanktoScope/PlanktoScope/blob/master/hardware/v2.5/case/PlanktoScope-Case.f3z)
2. try to open it with fusion360
3. it doesn't work

**Expected behavior**
I expected to be able to view the complete housing in fusion360.

**Additional context**
Add any other context about the problem here.

**Planktoscope version (please complete the following information):**
 - Raspberry version: [Pi4]
 - OS: [standard image]
 - Stepper motor hat [Adafruit]
 - Version [2.5]

Guelakais, 2024-06-11:
I have been working with the desktop version of Fusion360. Seriously, just upload the CAD files as .stl. This is then universal. You can simply export every component directly from Fusion as .stl. Then there should be far fewer problems.

ethanjli, 2024-06-11:
> Seriously, just

> You can simply

In my cultural frame of reference, people usually choose this kind of wording to 1) vent their frustration at the person they are talking to, and/or 2) express that they think they can do something better than the person they are talking to. Can you please clarify what you are trying to communicate with this choice of words, and can you please explain why you have chosen to use this tone with me?

> upload the CAD files as .stl

Providing STL file exports sounds like a reasonable idea to me. However, as a volunteer on the project who is not part of FairScope's paid staff and who only works on the PlanktoScope's community documentation site and the PlanktoScope's software (with a significant [backlog of tasks](https://github.com/PlanktoScope/PlanktoScope/issues/assigned/ethanjli) in addition to other volunteer work I do as a software maintainer on this project, such as triaging all new issues filed here), I currently don't have enough experience or interest in Autodesk Fusion 360 to learn how to use it to export each part of the CAD file to STL (my experience with CAD software is primarily OnShape and a bit of Solidworks); and as far as I'm aware, you are the first person to have submitted a request in the PlanktoScope community's spaces for STL exports of the CAD parts. Once we resolve your issues with the `.f3z` CAD file so that you can work with the CAD file, you're free to export the CAD parts to STL files and submit a Pull Request to add those STL exports this repo - and I'd be happy to help you get them integrated into the community documentation site (e.g. within https://docs.planktoscope.community/setup/hardware/v2.5/ ) so that we can help anyone in the future who also wants STL exports.

> I have been working with the desktop version of Fusion360.

Great, thanks for the information. On my end I've tried to reproduce the problem you reported, but I can't get the desktop version of Fusion 360 to work on my system (I run Linux but Fusion 360 is not available for Linux; when I try to launch Fusion 360 using WINE (via Bottles), Fusion 360 encounters a login error in the startup screen which prevents me from doing anything). I've made more progress in [the browser version of Fusion 360](https://www.autodesk.com/support/technical/article/caas/tsarticles/ts/4GAYwrd9SV4MJlh68wUATv.html): after I manually extract the `.f3d` files from the `.f3z` archive, upload them to Fusion's cloud file storage, and open the `fae88a10-4fb7-429c-b014-3a4d45e84809.f3d` file, I see the following result:

[image]

From what I can tell (given that this is only my first time using Fusion 360), all the parts load correctly except for the following linked external parts:

- C2 - Raspberry MicroSD v1:1
- C4 - PLANKTOSCOPE HAT V1 - FAN v3:1
- P4 - Linear stepper motor Robust version v2:1
- P4 - Linear stepper motor Robust version v2:2
- O1 - RASPBERRY PI HIGH QUALITY CAMERA v4:1

When you said in your initial report that "the unzipped f3d files could only be opened partially", I am wondering what opened and what did not open for you. Can you please provide a screenshot of what you see in your Fusion 360 desktop app when you try to open the `fae88a10-4fb7-429c-b014-3a4d45e84809.f3d` file? Also, are you able to use or try out the browser version of Fusion 360 for troubleshooting the problem you have reported?

closed 2024-06-11 by Guelakais

ethanjli, 2024-06-11:
@Guelakais If you managed to resolve the problem you reported here, can you please briefly summarize how you resolved the problem? This will help anyone in the future who encounters similar problems and finds this issue from searching. Thanks!