fairscope/PlanktoScope #472: Display MAC-Adress

Guelakais, 2024-09-16:
**Is your feature request related to a problem? Please describe.**
Sometimes you have to intervene deeply in the system architecture of the planktoscope from the outside or you simply want to register the planktoscope with the network administration.
**Describe the solution you'd like**
The following command displays the currently available MAC addresses of the device:
```sh
ip a
```
These would then only need to be displayed in the 'Wifi' or 'System' monitoring section. Since you are using a Node-Red dashboard, the following should already work.
The following command will dump all the mac-addresses of the device in to a .txt file in the `$HOME` folder of your System:
```sh
ip a > $HOME/mac_addresses.txt
```
Now you can read it in inside the `NODE-RED` Dashboard:
```javascript
const fileSelector = document.getElementById('$HOME/mac_addresses.txt');
  fileSelector.addEventListener('change', (event) => {
    const fileList = event.target.files;
    console.log(fileList);
  });
```
Now you can express it in the UI.
**Describe alternatives you've considered**
Make a dedicated ssh access point like:
```sh
ssh device_url
```

ethanjli, 2024-09-18:
Thanks for starting this feature request! I agree with the overall request, and I think we can solve this use-case half-way without any further decision-making. However, having full integration into the Node-RED dashboard will require me to discuss this feature request with the people who are redesigning the Node-RED dashboard. Here is my full analysis:

### Motivation

I agree that it would be useful to give users a way to view the device's MAC addresses through a GUI, without having to run a terminal command. My reasoning is that there are some university networks - such as the one at my university - which require entering a device's ethernet and/or Wi-Fi adapter MAC addresses in a device registration form for getting internet access for that device on those networks.

### User-Facing Design

If we decide to display MAC addresses in the the current version of the Node-RED dashboard (which is version 2 of the dashboard), we could add it to either the "System Monitoring" page or the "Wifi" page. It would probably be the latter page. However, the current version of the Node-RED dashboard is going into maintenance mode with no new features to be added; see https://github.com/PlanktoScope/PlanktoScope/blob/master/software/CHANGELOG.md#deprecated-2 for details. It will be replaced by a complete redesign/rewrite of the Node-RED dashboard (which will be called version 3 of the dashboard). 

In the current prototype of the new rewrite/redesign, the "Wifi" page appears to have been removed, while the "Monitoring" page has been reduced in scope to only show CPU/RAM/disk usage elements previously in the "Administration" page. Other system information has been split between the Home page (which shows the PlanktoScope's model, serial number, machine name, software version, disk usage, etc.) and the Hardware Settings page (which shows the Raspberry Pi's amount of RAM, as a read-only field; but this page's scope is more about things like the type of HAT, types of lenses, hardware version, etc.).  I don't know whether we would want to display MAC addresses on the Home page (probably too much technical detail for the home page) or the Hardware Settings page (whose scope currently does not really overlap with things like MAC addresses). This decision seems a bit tricky to me, but it is not my decision to make; it will require some discussion. Also, the rewrite will probably not be ready within the next two months.

### Alternatives

In the meantime, a very simple and easy-to-implement-immediately partial solution would be to auto-generate a file (e.g. `/run/mac-addresses`) during startup which can be viewed via the system file manager (e.g. at http://planktoscope.local/admin/fs/files/run/mac-addresses) which lists the MAC addresses. We could then link to this file from the landing page (or even just display the MAC addresses directly on the landing page), if we wanted to make it easy for users to see the MAC addresses even before the Node-RED dashboard adds a place for that information. This change is small enough for me to implement in the next software release after v2024.0.0.

### Implementation details

The MAC addresses are represented by the kernel's system information virtual filesystem, at `/sys/class/net/{interface name}/address`. So we don't need to run `ip a` to query addresses, instead we can just read the contents of `/sys/class/net/eth0/address` and `/sys/class/net/wlan0/address` (or, if we are interested in the MAC addresses of other network interfaces, we can iterate through the subdirectories of `/sys/class/net`). This way, we can have the Node-RED dashboard (or the PlanktoScope's landing page) directly read a few specific files to determine MAC addresses, without having Node-RED run shell commands (which is something we should stop doing, as part of containerizing the Node-RED dashboard to enable easier/simpler/more-robust software updates on the PlanktoScope as well as a future path to more thorough software testing of the Node-RED dashboard).

This also means that we can run a script during boot to assemble these MAC addresses into a single file, without having to write a more sophisticated regex correctly to parse the output of `ip a`. For example, this command will generate a valid YAML file:
```
( cd /sys/class/net/ && for i in *; do printf '"%s": "%s"\n' "$i" $(cat "$i"/address); done ) > /run/mac-addresses.yml
```
with the following example of the contents of `/run/mac-addresses.yml`:
```
"br-118af865b414": "02:42:33:ea:db:7c"
"br-c11607a0a609": "02:42:46:9b:64:13"
"br-d09ec6f6d85c": "02:42:06:fa:3c:b3"
"br-f2d182b51e21": "02:42:20:36:d6:01"
"docker0": "02:42:e1:ca:33:bd"
"eth0": "dc:a6:32:4f:51:ef"
"lo": "00:00:00:00:00:00"
"veth07c3199": "2e:a8:c6:f2:1f:e4"
"veth156605d": "ca:9b:fd:ae:7e:b0"
"veth1742758": "46:dc:40:a1:d9:ad"
"veth183233f": "26:d6:7e:cc:fd:d0"
"veth258597d": "2a:85:dd:b6:40:db"
"veth43e5e2a": "56:7d:a8:5a:0a:31"
"veth512d96a": "3e:c5:19:48:ea:7f"
"veth5202129": "52:e2:1e:3a:c0:09"
"veth5449344": "3a:97:c0:e0:dd:85"
"veth6289ee7": "ce:50:fc:76:6a:33"
"veth63181f4": "0a:ba:b9:d6:11:ee"
"veth82072a5": "c6:4e:b8:d3:a6:09"
"vetha88edda": "12:8f:2f:7b:14:9e"
"vethb0205b5": "fa:c2:4f:55:62:98"
"vethd013de0": "c6:5a:01:b4:67:fc"
"vethd582bb4": "ba:71:a1:c1:f4:e3"
"vethdd24bb9": "5e:14:d8:e6:3c:4e"
"vethde1c812": "72:53:35:da:af:63"
"wlan0": "dc:a6:32:4f:51:f0"
```

This is probably what I'd do for my proposed partial solution "in the meantime" before adding MAC addresses into the GUI design of the Node-RED dashboard.

ethanjli, 2025-01-18:
https://github.com/PlanktoScope/pallet-standard/pull/70 (which is integrated into the OS build scripts by #522, along with updates to the [developer preview version of the documentation site](https://docs-edge.planktoscope.community/)) fixes this issue (though not exactly in the way originally requested) by adding a link (named "MAC address viewer") on the landing page to a file viewer for a YAML file which lists the MAC addresses of all network interfaces on the Raspberry Pi. If you're running the v2024.0.0 software, I think it should be possible for you to try out this fix on an internet-connected PlanktoScope by following the [in-place upgrade guide](https://docs.planktoscope.community/operation/software-upgrades/#perform-an-in-place-upgradedowngrade) with the following upgrade command:

```
forklift pallet switch github.com/PlanktoScope/pallet-standard@1a260ab
```

If you need help getting your PlanktoScope connected to the internet, please refer to the relatively new instructions on the documentation site's [networking operations guide](https://docs.planktoscope.community/operation/networking/#connect-your-planktoscope-to-the-internet) - there may be some problems with those instructions, so it would be very helpful for me if you could try it out and list anything which seems confusing, inaccurate, or missing.

The file reporting MAC addresses is generated in the way I had described/planned in https://github.com/PlanktoScope/PlanktoScope/issues/472#issuecomment-2359193478 . That file is automatically updated every two minutes.

For any remaining requests or problems on this topic, please feel free to add more comments to this issue and either I can follow up on this issue or I can make a new issue.

PR #522 (2025-01-18): Report MAC addresses
This PR resolves #472 by integrating https://github.com/PlanktoScope/pallet-standard/pull/70 and updating the technical reference docs and updating the operations docs to reference the added functionality.
closed 2025-01-18 by ethanjli