fairscope/PlanktoScope #474: EEPROM listed in BOM is deprecated; new versions available

ceeller, 2024-09-17:
https://github.com/PlanktoScope/PlanktoScope/blob/b785ad8c71e9ff4ab3ed1e045963fc4809d0051a/hardware/v2.5/hat/assembly/Planktoscope-Hat-bom.csv#L27

The M24C32-WMN6 EEPROM listed on line 27 is no longer manufactured or is very difficult to find. There are two updated versions available:

- [M24C32-WMN6TP](https://www.mouser.com/ProductDetail/STMicroelectronics/M24C32-WMN6TP?qs=%252B9pWl1iD4MWFKTqSdEAgrw%3D%3D)
- [M24C32-WMN6P](https://www.mouser.com/ProductDetail/STMicroelectronics/M24C32-WMN6P?qs=tZUCD5p2VWo7wmvQ2UUgjQ%3D%3D)

Are either of these chips suitable for use in the HAT?

ethanjli, 2024-09-18:
Here are some relevant messages copied from the PlanktoScope Slack workspace:

- https://planktoscope.slack.com/archives/C01V24TUQ13/p1720566483312979 from Kazi Uzayr Razin:
  > Hi everyone, my PCB provider ran into some issues and suggested alternatives. This is what he said:
Line 24: Original MPN: FM4007W - Could not locate. Suggested Alternate MPN: FM4007-W. Please approve, provide alternate MPN, DNP or Consign.
Line 26: Original MPN: M24C32-WMN6 - Unavailable. Suggested Alternate MPN: M24C32-WMN6P. Please approve, provide alternate MPN, DNP or Consign.
Line 27: Original MPN: PCF85363ATL/A - Unavailable. Suggested Alternate MPN: PCF85363ATL/AX. Please approve, provide alternate MPN, DNP or Consign.
Should I confirm his suggestions?
- reply in thread from me:
  > At first glance, these suggestions seem reasonable:
  > - Line 24 (rectifier diode): I'm not familiar with this part. From reviewing the datasheet at https://www.rectron.com/public/product_datasheets/fm4001w-fm4007w.pdf, I would've expected this substitution to be to FM4007W-W  (where the -W suffix just indicates a packaging format). But there doesn't seem to be any part named FM4007(_) where (_) is not W - so FM4007-W probably does refer to the same part as what the HAT requires.
  > - Line 26 (EEPROM): From reviewing the datasheet at https://www.st.com/en/memories/m24c32-w.html, I think replacing M24C32-WMN6 with M24C32-WMN6P will just add additional environmental compliance guarantees to the parts (e.g. free of certain flame retardants)
  > - Line 27 (RTC): From reviewing the datasheet at https://www.nxp.com/docs/en/data-sheet/PCF85363A.pdf , I think PCF85363ATL/AX is the orderable part number for PCF85363ATL/A , which is not an orderable part number. So this substitution should not involve any change of what you would actually receive, it is just a correction for what your PCB provider needs for ordering the part.
- as a follow-up, in https://planktoscope.slack.com/archives/C01V24TUQ13/p1722356805142039 Kazi reported some issues from the PCB assembler about incorrect footprints of parts in populating the PCB with components related to the hardware RTC (see #475 for additional Slack messages relevant to that). However, the PCB assembler did not report problems with the EEPROM chip, so presumably it was successfully placed onto the board.

ethanjli, 2024-09-18:
I think we should treat this issue as a potential typo/error in the BOM, so I will keep this as an Issue instead of converting it to a Discussion. If/when we get confirmation that the EEPROM can be driven correctly even if we substitute M24C32-WMN6 with M24C32-WMN6P, then we should change the listed part number in this repository's v2.5 BOM, and then we can mark this issue as fixed.

Note: currently the PlanktoScope software does not use the EEPROM; such software functionality is stalled (due to lack of people to work on it and lack of priority compared to other software functionalities) in the [design/prototyping phase](https://github.com/PlanktoScope/proposals/pull/3/files).

ceeller, 2024-09-19:
Thank you. I have updated my local copy of the BOM to list the manufacturer as STMicroelectronics and the part number as M24C32-WMN6P.

closed 2026-11-09 by chevreuill3000